SWC-Advanced-Microscopy/StitchIt #14: Revisit the way we subtract the background tiles at some point

raacampbell, 2016-11-22:
This is perhaps a bit ad-hoc?

raacampbell, 2017-03-13:
Thinking of addressing this with a background subtracter class that contains the images and the methods to apply the correction. Then can have different classes for different correction types. e.g. for average tiles and for cidre.

raacampbell, 2017-03-14:
Related to this is the now orphan `lowValueThreshold` parameter in the INI file. If we indeed calculate this value adaptively, then lowValueThreshold (which is currently doing nothing) should definitely be removed.

raacampbell, 2017-05-11:
I think this has been done?

raacampbell, 2017-08-03:
Try mixture of Gaussians approach for getting the offset (was #61)

raacampbell, 2017-08-03:
Upgraded because of [BakingTray Issue 57](https://github.com/BaselLaserMouse/BakingTray/issues/57) which caused channel 2 of MoA_20 to stitch badly.

raacampbell, 2017-08-04:
We now (c553da5209a3b6c770af25353dcfd804f009c42c) calculate the offset based on a Gaussian mixture model applied to the dimmest tiles. This allows for the use of a biased amplifier and also corrects for acquisitions where the auto bias adjustment in ScanImage is causing problems. However, there is no ability currently to enable/disable the application of the offset when tiles are loaded. It is currently done by editing a flag in `bakingtray.tileLoad`. We need maybe an INI setting to deal with this if it becomes a feature we really need to use.

ablot, 2017-11-02:
I think that most of this issue has been fixed on the dev branch.

Related: I'd like to add an option to fix the offset in `preProcessTile` (for instance calculate it on the first optical plane and then keep it constant). I think it would solve most of the issue if the offset is constant during the recording. Do you have any objection/advice?

raacampbell, 2017-11-02:
Sure, let's do that and see how we go. It ought to work.

closed 2017-11-02 by raacampbell