SWC-Advanced-Microscopy/StitchIt #75: Missing tileStats.mat causes unclear error message

raacampbell, 2017-10-05:
@ablot had a problem with the `tileStats`, more precisely `tileStats.offsetMean`. It had only 2 values, 0 for channel 1 that I did not record and an offset for chan 2. Nothing for chan 3. Then stuff does not stitch. So we should:

* [ ] Make it possible to generate the tile stats for a single channel only by saving a separate `tileStats.mat` for each channel.  
* [ ] Fail gracefully is the tile stats are missing.

@ablot, does that cover it?

ablot, 2017-10-05:
That should cover it. 
`tileLoad` already fails quite gracefully if the `tileStats` file is missing. It's less elegant if the file is incomplete (my case). In any case, the error will not be exposed by `preProcessTiles`, you just have a "Cannot load channel ...." message (but that should be enough I guess)

For now the solution to break the loop is to set  `doOffsetCorrection=0` in `tileLoad`, run `preProcessTiles`, revert `doOffsetCorrection=0` and run again `preProcessTiles`.
It seems to work so far.

ablot, 2017-10-05:
If I start `stitchAllChannels` on a clean new folder with only raw data, it raises plenty of warning of `tileLoad` trying to offset when there is no `tileStats` but seems to do most of the stitching correctly. The illumation correction was not done, but I'll see if it is an unrelated problem or not once I get the offsets right. However `stitchAllChannel` doesn't write any `tileStats`file.

If I do `preProcessTiles` with `-1` as tile, it complains that the `tileStats` don't exist (since `stitchAllChannel` did not create them) and then it creates them. But if I re-run `preProcessTiles`, it doesn't complain and create a new `tileStats`. I'm afraid that if I do that, it's going to calculate the offset on an offset-subtracted dataset and fuck up all the `tileStats` files. Am I correct in thinking so?

raacampbell, 2017-10-05:
Yes. So we should ensure that `preProcessTiles` handles the scenario where the `tileStats` already exists.

raacampbell, 2017-10-05:
@ablot I'll start working on this on the `dev` branch.

ablot, 2017-10-05:
I think that for now, my first run of `preProcessTiles` creates the offset but does the `calcAverageMatFiles` (and the `writeComCorCoefs` but I don't use that) with data that is not offset subtracted. That might cause problem in later steps.

annoying problem: if I run `preProcessingTiles(-1, 0, (2:4))` once to create the tileStats, delete the average folder and then run `preProcessingTiles(0, 0, (2:4))` to use the correct offset and generate the averages, the `calcAverageMatFile` fails.
I think it's because the `lowValue` read from `tileStats` is *not* offset subtracted... so all my images are below `lowValue`

raacampbell, 2017-10-05:
How about we do this:

1. `preProcessTiles` always load the image stack without the offset subtracted (even if `tileStats` exists). 
2. Then calculates the `tileStats`
3. Then it applies the offset it calculates. 
4. Profit.

ablot, 2017-10-05:
sounds good

raacampbell, 2017-10-05:
I think `tileLoad` it was indeed failing gracefully when a `tileStats` file does not exist. @ablot fixed this. So I check that off the list.

raacampbell, 2017-10-05:
`preProcessTiles` uses the offset subtracted data for average images - 0e6f1903d3b0b3114727638f5c64cec2a0228ab5

raacampbell, 2017-10-05:
@ablot I tested this quickly and it seems to work. Can you see if it resolves your issues? Checkout `dev`.

raacampbell, 2017-10-05:
I think the issue you mentioned with the negative numbers is still there. I didn't subtract the offset before calculating that stuff. So that's still to do. This is because I updated the image data at the end of [writeTileStats](https://github.com/BaselLaserMouse/StitchIt/blob/dev/code/preProcessTiles/private/writeTileStats.m) I think what you were getting at is that it should happen before the means and SDs are calculated, yes?

ablot, 2017-10-06:
Didn't you want to keep everything in tileStats to be calculated in non-offset subtracted data? Then there is no need to change. I think there were two different problems
1 - I haven't changed `calcAverageMatFiles` to subtract the offset on `lowValue` (that could also be done in `preProcessTiles` when calling `calcAverageMatFiles`)
2 -For now `preProcessTiles` should be run once when `tileStats` doesn't exists. I tried to re-run it after deleting the 'averages' folder and that fails for multiple reasons (#79 but also the fact that `imStack` is modified by `writeTileStats`)

Let me know if you want `calcAverageMatFiles` to subtract the offset from `lowValue` himself or if that should be done upstream and I'll adapt the code and try again.

raacampbell, 2017-10-06:
@ablot I think this is now done. Is that right?

raacampbell, 2017-10-09:
I merge into `prerelease` -- 787e3072a428a1ab5e41ea1f36063cddf67beab8 and if all is good [will merge into a master in a little bit](https://github.com/BaselLaserMouse/StitchIt/wiki/Contributing-code-to-StitchIt).

closed 2017-10-10 by raacampbell