SWC-Advanced-Microscopy/StitchIt #78: Have a separate folder for processed data

ablot, 2017-10-05:
For now, processed data that is used to generate the stitched images is stored in the `RawData` folder. It would be nice to move that to a separate folder (`ProcessedData` or something like that`) that can be deleted without fear of loosing anything.

raacampbell, 2017-10-09:
We'll leave the `tileIndex` file where it is. It's only needed for TissueCyte data anyway, since BakingTray saves the tile locations at acquisition time.

raacampbell, 2017-10-09:
@ablot I have something (0140d15058dc8b1e21f02675cd3a9dada47a8482) that seems to work so far (tested quickly once) and produces this:

```
>> ls -l  rawData
total 0
drwxrwx---  139 rob  _lpoperator  4726  9 Oct 14:07 YH_312-0001
drwxrwx---  139 rob  _lpoperator  4726  9 Oct 14:07 YH_312-0002
drwxrwx---  139 rob  _lpoperator  4726  9 Oct 14:07 YH_312-0003
drwxrwx---  139 rob  _lpoperator  4726  9 Oct 14:07 YH_312-0004
drwxr-xr-x    5 rob  _lpoperator   170  9 Oct 14:08 averageDir
drwxr-xr-x    6 rob  _lpoperator   204  9 Oct 13:44 stitchitPreProcessingFiles

>> ls rawData/stitchitPreProcessingFiles/
YH_312-0001 YH_312-0002 YH_312-0003 YH_312-0004

>> ls rawData/stitchitPreProcessingFiles/YH_312-0001/
averages      tileStats.mat
```

ablot, 2017-10-12:
seems to work so far. I just needed to update loadtile to find the proper tilestats.

raacampbell, 2017-10-16:
Works. Need to do the docs and merge into master.

raacampbell, 2017-12-13:
We've been using this for some time and it works well.

closed 2017-12-13 by raacampbell