SWC-Advanced-Microscopy/StitchIt #80: calcAverageMatFiles fails if there is a plane with almost no data

ablot, 2017-10-06:
The few last Z plane of my last acquistion had basically no data (outside of the slice). `calcAverageMatFiles` ignores successfully cases with no data at all (if more than 85 % of the tiles are below `lowValue` but crashes if there are less than 85% rejected and the remaining tiles are all on even or uneven raw. (L87 or 103)

raacampbell, 2017-10-06:
I did see it fail with your 800 nm data set. Does this commit fix it? bd49c0e75b6c900234ee31baf53a0c83ec9bc872 It now runs on that dataset, but I've not looked at the stitched images yet.

ablot, 2017-10-06:
That works for the dataset at 800nm indeed, but I think it could still happen. See #81 for a potential lazy fix (just skip the annoying situation)

raacampbell, 2017-10-06:
Fixed by [pull request](https://github.com/BaselLaserMouse/StitchIt/pull/81)

closed 2017-10-06 by raacampbell