SWC-Advanced-Microscopy/StitchIt #83: Generate tileStats.mat one channel at a time

ablot, 2017-10-10:
Suggestion of feature:
- make a `tileStats.mat` file per channel.
- have an optional argument after `illumCorrection` to tell which channels needs to have their `tileStats` created but no average.

For now calling `preProcessTiles` reads or generate the `tileStats.mat`. If I process one channel and try to run `preProcessTiles` with another channel later, things are likely to go wrong. Having a file per channel and generating only channels that do not exist yet would be nice.
My current use case: pre processing when the data of one physical slice doesn't fit in memory (too many optical planes). Then doing it channel by channel might help. 

If you think it's a good idea I can start working on that.

raacampbell, 2017-10-10:
Yes, this is a good idea. See also Issue #68 and Issue #20. I think all of of the input arguments should be parameter/value pairs since we never do the comb correction, for instance. 

Go for it!

raacampbell, 2017-10-11:
[merged pull request](https://github.com/BaselLaserMouse/StitchIt/pull/85) and will test on the next brain.

raacampbell, 2017-10-11:
I see the following error: 
```
tileLoad: No tile index file: rawData/singRabies_01-0002/tileIndex
Creating stats file: rawData/stitchitPreProcessingFiles/singRabies_01-0002/tileStats_ch2.mat
Reference to non-existent field 'offsetMean'.

Error in writeTileStats (line 122)
    offsetMu = mean(tileStats.offsetMean(1,:)); %since all depths will have the same underlying value

Error in preProcessTiles (line 238)
        [tileStats,~] = writeTileStats(imStack(chan,:), tileIndex(chan,:), sectionStatsDirName, statsFile);

Error in syncAndCrunch (line 334)
  analysesPerformed = preProcessTiles(0,combCorChans,illumChans,0); %PRE-PROCESS TILES
```

ablot, 2017-10-11:
Ok. I managed to replicate (just rename `tileIndex`) and fix.
With  #86, if no data is loaded, the section is skipped.

raacampbell, 2017-10-12:
If I acquire channels 1 and 3 and incorrectly tell preProcessTiles to do illumination correction for channels 1, 2, and 3 then it will create tileStats_ch2.mat even though it can't ever fill it with data.
(EDIT: In fact this is also a bug in the prerelease branch. I'll work on that )

Also: I think if not all tiles are present (during syncAndCrunch) then it makes a mostly empty tileStats file with no data in it and then it crashes when it later attempts to load this. Can you look into this, @ablot ?

raacampbell, 2017-10-12:
Fix missing channels (e.g. user recorded chans 1 and 3 but not 2) bug in `prerelease` branch, but this does not handle @ablot code for different `tileStats` files per channel. It just allows me to carry on with my current recording. 4e700ec4643bb5bd4cc7c277b0733092ef130b11

raacampbell, 2017-10-15:
@ablot  Do you have time next week to look at the bug from three days ago?

ablot, 2017-10-16:
I'll have a look today

raacampbell, 2017-10-16:
There are some bug fixes that are in `prerelease` that aren't yet in `dev`. So we'll need to merge `prerelease` into `dev` and test.  Issue #88 in particular.

raacampbell, 2017-10-16:
The above changes are safe and good to add to `dev` but this bugfix maybe isn't 4e700ec4643bb5bd4cc7c277b0733092ef130b11 so I'll leave it out for now. We'll need to decide only when it's time to merge into `prerelease`.

raacampbell, 2017-10-16:
Putative bugfix for incomplete `tileStats.mat`: 962d617e0733bf62fe1a190d37bb0260011d07e3 I return a loadError=true from `load_imstack` if the `imStack` array is empty. I will test on the next sample.

raacampbell, 2017-10-16:
Done 1b1c6c7b1b86f2f1c6c62dd4fbd32f308d78a708

closed 2017-10-16 by raacampbell

ablot, 2017-10-16:
You've been faster than me!
I'll keep my version on the side for now. 
https://github.com/ablot/StitchIt/commit/ab48be45702676e7c8e460e70e3945eb21ff48c6

I'll try to merge and PR if we run into other problems (I think the current version will create empty subfolders in the average folder for channel that don't exists if they are in the 'illumCorrection` argument) or if loading more than one channel at a time starts to be a problem for the RAM.