SWC-Advanced-Microscopy/StitchIt #89: I see new messages saying "Skipping illumination correction" during syncAndCrunch

raacampbell, 2017-10-17:
I don't recognize these. Maybe they are just new messages from the new `preProcessTiles`, but should check to make sure it's not skipping stuff it should be doing.

ablot, 2017-10-21:
It should be printed if a slice for which the average have already been generated is going through `preProcessTiles` again. But it just checks:
```
length(sectionsToProcess)==1 && sectionsToProcess==0 && exist(aveDir,'dir') && all(cellfun(@exist, chanDirs))
```

Maybe it should be clearer. "Average folder already exists. Skipping illumination correction" ?

ablot, 2017-11-16:
Or should I just not display anything for section already preProcessed?

raacampbell, 2017-11-16:
No harm displaying a message.

closed 2017-11-16 by ablot