SWC-Advanced-Microscopy/StitchIt #90: Bin raw data before crunching

ablot, 2017-10-21:
I'd like to bin the images 2x2 before processing them. I've seen in #20 that you might want to change the arguments of `preProcessTiles`. What did you have in mind? Should I just add a last (optional) argument like `binSize`, do the binning before creating the `tileStats` and use only the binned data for everything else? Or would you rather have something more outside of `stitchIt`. I bin the data myself and then trick `stitchIt` into using them by changing the directory names?

raacampbell, 2017-10-21:
I'm not sure what I have in mind. Somehow I want to make pre-processing functions that can be chained together, but I haven't though how. 

Regarding down-sampling, I'd say we should start off with a stand-alone function. [This is the TIFF writer example](https://bitbucket.org/snippets/raacampbell/xxobX/petrs-int16-tiff-writer-for-matlab)

ablot, 2017-10-23:
My new binned tiffed were saved without the scanimage header, so `tileLoad` fails to read them. Should I copy the whole `Software` field in my tiffs?

raacampbell, 2017-10-23:
Yeah, we'll need those. Obviously we won't be able to test this for TissueCyte data, so just make a note of that in the docs.

ablot, 2017-10-23:
I haven't edited any of the TissueCyte function. I remember at least changing `bakingTray/tileLoad` to find the proper `tileStats`

raacampbell, 2017-10-23:
Yeah, don't worry about that. We're never going to use their acquisition software (Orchestrator) again. In a year or so I'll probably stop supporting it entirely.

ablot, 2017-10-24:
Ok. I made a PR with the first version. It seems to bin and to produce something that `preProcessTile` can use. I'll test a bit more, you can wait to merge and suggest changes if you have any idea.

- `binNdArray` is a bit stupid and works dimension per dimension (that means no median binning). 
- I cannot bin raw data in Z since the channels are interleaved (At least it's not implemented for now).
- I'll add something to run the `binNdArray` on stitched image. That would be useful for me to do max projection over ~10 slices on Z

raacampbell, 2017-10-24:
I wouldn't bin in Z. It will produce bad results unless the sampling is very fine. You can try `resampleVolume` to bin stitched images.

ablot, 2017-10-24:
Binning in a new directory works fine but one need to manually copy the `tilePosition.mat` for the rest of the processing to work.

- [ ] Check that `tilePositon.mat ` is not affected by the binning
- [ ] make  a function to copy all the tilePosition files to the new target (maybe as optional argument of `binRawData`

After data, changing the raw data directory in the ini file should be enough

ablot, 2018-04-23:
It's been working for a while now. I'm closing this. There is room for improvement but I'll open another issue if we decide to work more on that.

closed 2018-04-23 by ablot