SWC-Advanced-Microscopy/StitchIt #99: Use ScanImage offset value for subtraction

raacampbell, 2017-11-06:
I suggest we continue to calculate the offset for the moment, in case the ScanImage value is for reason problematic. In the mean time I will do the following:

* [ ] Enforce the automatic offset calculation by BakingTray, so the user can not switch it off. 
* [ ] Have `tileLoad` use either the single value in `scanSettings.mat` or the value from each tile (the user will select). We will default to the `scanSettings.mat` value to have a single value for all recordings.
* [ ] Confirm what happens if the user selects offset subtraction in ScanImage.

raacampbell, 2017-11-06:
If "Subtract Offset" is ticked then the file is *saved* with the offset subtracted. So this is most likely bad and we don't want this. 

BakingTray now enables the offset auto read each time it is armed.

ablot, 2017-11-16:
#103 is doing the default behaviour. I haven't added the option to read offset from the current image. Would that be in the ini file?

raacampbell, 2017-11-16:
How about we don't add it as an option. If we see problems and we need it sometimes then we make the option. The fewer options the better.

raacampbell, 2017-12-13:
@ablot  is this done?

ablot, 2017-12-13:
Yes. 
https://github.com/BaselLaserMouse/StitchIt/commit/fc5721811c8c8cf057ee82de7af982127bb609b9

ablot, 2018-04-23:
Currently, scanimage offset of the first tiff file is used by `tileLoad` when open the data but `preProcessTile` still generates the `tileStats` with the GMM fit of dim tiles. This value is used only by `calcAverageMatFiles` but that means that the offset of the average is different from that of the data.

I'm currently testing if using scanimage for everything solves my problem.

ablot, 2018-04-23:
Update: The problem of bright average with RR brains is solved by using scanimage offset everywhere.
My current quick fix just writes in `tileStats` two offset: offset GMM that is never used and an offset from scanimage which is the exact same value for every single frame. `calcAverageMatFiles` uses this scanimage offset. Then `tileLoad` always re-read scanimage offset from disk. That's obviously all cumbersome and stupid.

What do you want to do @raacampbell ? Should we keep on using the first scanimage offset? Should we not use SI but calculate one offset on the whole dataset somehow? Or do we really need something tracking offset during acquisition (so a non-constant offset)?