SWC-Advanced-Microscopy/StitchIt #104: generate_MaSIV_list fails silently with non numeric channel names

ablot, 2017-11-16:
That means that the output of mergeChannels needs to be renamed to an arbitrary number. 

- [ ] Check that masiv itself can handle arbitrary channel names
- [ ]  If it cannot, change the regexp in generate_MaSIV_list to do only numeric only channel
- [ ]  if it can, change the num2str and downward code to handle text.

Low priority since renaming the folder works fine

ablot, 2017-11-16:
More precisions on the current behaviour:

Folders with a number in their name are used by `generate_MaSIV_list` but if `str2num(folderName)` fails, then the channel shows as `Ch` in masiv. Masiv can open this channel perfectly fine. So I guess I can tick the first box.

raacampbell, 2017-11-16:
Yeah, I think we should try to get MaSIV to do that. Arbitrary channel numbers are annoying. Later you don't know what they are.

ablot, 2017-11-16:
Done. Tested on linux. I think that I've replaced all the `[dir, '/', dir]` by `fullfile` except that used to write in the yml file. Hopefully that should solve #29 and still generate a file list with unix separators.

closed 2017-11-16 by ablot