SWC-Advanced-Microscopy/StitchIt #113: recipe files have parameters defined multiple times

ablot, 2018-05-21:
In the recipe file pixel size and number of line/rows are defined twice. Should they be merged or do you have a reason to keep 2 versions?

Example extract of recipe file:
```
Tile: {nRows: 566.0, nColumns: 566.0}
VoxelSize: {X: 1.668, Y: 1.668, Z: 6.666666666666667}
ScannerSettings:
  pixelsPerLine: 566.0
  linesPerFrame: 566.0
  micronsBetweenOpticalPlanes: 6.666666666666667
  micronsPerPixel_cols: 1.668
  micronsPerPixel_rows: 1.668
```

ablot, 2018-05-21:
So I can ignore all the `ScannerSettings` for the rest of the processing and treat that as log? (when I downsample data or adjust pixel size to help to stitch)

raacampbell, 2018-05-21:
Pretty much, although that scanner pixel size is what was used to decide
where the stage should move.

Rob

closed 2018-05-21 by ablot