SWC-Advanced-Microscopy/StitchIt #194: Matlab error on Ubuntu

ederancz, 2021-08-31:
MATLAB R2018a, Ubuntu 20.04. 
stitchAllChannels crashes after averaging tiles, right after creating empty stitched data directory tree. 
Error in imDistCorrect (line 193). 
Can't reload 'usr/local/MATLAB/R2018a/bin/glnxa64/libmwcoder_types.so'

The required toolboxes are installed, but it looks like some of the runtime libraries may be missing somehow. 
Did anyone run into this before?

raacampbell, 2021-09-01:
That's weird. May be a problem with your install but also that's an old version of MATLAB. Try upgrading it and see if that helps. I'm running 2020a right now but it's not necessary to be quite that up to date.

ederancz, 2021-09-01:
I was hoping to get around without sudo rights. 
It works with a new install of 2021a.

closed 2021-09-01 by ederancz