SWC-Advanced-Microscopy/StitchIt #203: syncAndCrunch fails to generate preview (slow connection?)

mkrumin, 2022-11-25:
We have a slow connection between the `ScanImage` and the processing machines, and for high-res multi-channel acquisitions `rsync` is lagging behind acquisition and the lag only increases as acquisition progresses.
Watching `syncandcrunch` command window and `forWWW` folder suggests that it only generates web preview when all the data is synced to the processing machine? The message says that it generates the preview, but the folder is always empty.

raacampbell, 2022-11-25:
This is associated with Issue #202 

It does only try to generate the preview once the initial rsync is done, yes. Further, it currently waits until one more section has been acquired before starting. I am changing the latter, so it will build a preview image immediately so long as at least one section is present (3d1f63e287a4f2bc37254bcfebbeb8a99edc454d).

Ultimately, though, your connection is very slow and fixing that is the priority.

mkrumin, 2022-11-25:
Agree!

raacampbell, 2023-01-19:
I also had this issue today: A PC got stuck at 100 MB/s and no previews ever appear.